obra/kicad-tools #11: Uploading docker image to DockerHub

nnarain, 2019-10-15:
Hey!

Basically my use case for `kicad-tools` is to auto-generated schematic pdfs, BOM, and gerber files on a git tag (Currently using Github Actions) and upload to the Github Release page. Which I have a working example of [here](https://github.com/nnarain/esp-light-controller/commit/60d54203bdc93cd723be8fda18e389d0c8755e9b/checks?check_suite_id=243343868).

However, building the docker container takes quite a while. You can see [here](https://github.com/nnarain/esp-light-controller/blob/develop/.github/workflows/release.yaml#L19-L23) that I've uploaded a docker image based on my fork of `kicad-tools` to Dockerhub to get everything moving along smoothly.

Aside from the CI use case, it would make installation easier for user as well (no need to build the image).

Thoughts?

obra, 2019-10-15:
I certainly don’t object. I just don’t know much about docker hub. what do i need to do?.

nnarain, 2019-10-15:
I'm rather new to docker as well. Setting up the automatic image build was pretty straight forward though. 

1. Create a docker hub account (and link github account)
2. Create a new repository (on dockerhub).
3. Under the build tab, configure the automatic build. Set the source repo (`obra/kicad-tools`). The default build rule should be fine (source branch is master, and the tag name defaults to latest).

A build will trigger once complete to generate the initial image. Afterwards, any push to `master` on `obra/kicad-tools` should trigger a new docker image build (tagged as `latest`).

rs-development, 2019-11-16:
I would love to see this as well, no hassle to build the image local. Its alot easier to automate it with CI jobs.

rs-development, 2019-11-16:
So I have tried your work and it is awesome! 

I think it would be a perfect suit if this could be run in a CI job. Normally we use a Docker CI, which is running an image and doing things inside. But since your magic is already inside a docker container i am not sure how to do it.

Maybe someone has a clever idea?

obra, 2020-10-23:
Solved with #25 by @charliebruce

closed 2020-10-23 by obra