obra/kicad-tools #14: Where should python scripts live?

adamwolf, 2019-12-18:
Hi @obra, thanks for your great work.

I am working on putting some python together that tells KiCad to fill zones in a PCB.  Where should that live? in your upstream kicad-automation-scripts?  In the upstream-upstream kicad-automation-scripts?

Are those just for xdotool scripts? Do we want yet another upstream?

Let me know, I want to spend some hours on this over the next few weeks.

obra, 2019-12-19:
in my upstream isn't wrong. but also breaking that link and moving
everything from my upstream into this repo is also fine.  I don't think
we get a lot of values out of having a bunch of tiny repos.

adamwolf, 2019-12-20:
Is there a good way to bring the upstream into this repository and preserve history?  I am comfortable in git but I'm not sure I've had to do this before.