obra/kicad-tools #29: Add JLCPCB-specific fabrication outputs

kroesche, 2021-09-23:
As discussed here: https://github.com/obra/kicad-tools/issues/27

I will work on a PR for this feature. I created this issue to have a place to track it and have any further discussion about it.

conversation summary from obra:

_"I'd also love a PR to add explicit generation steps for jlcpcb (and any other pcb house), maybe integrating https://github.com/matthewlai/JLCKicadTools for JLC."_

AND

_"I think that in my ideal world, everything would go into the docker container. That's part of the point of having Docker there at all - getting a known-good environment that should work cross-platform._

_What I'm envisioning is a new makefile target of something like_

    make fabrication-output-jlcpcb

_that would create the zipfile, the BOM and the placement file."_

kroesche, 2021-09-24:
I am working on this one first, and then I will look at #30.

kroesche, 2021-09-29:
@obra I pretty much have completed this work in my own branch. However, I realize that there is still a file that I have to manually generate before you can run the new stuff to generate the JLCPCB parts files. Per the JLCKicadTools, you first have to use pcbnew to generate a footprint position file.

Up to now, for my own projects, this is what I have done. I generate the position file using pcbnew UI. Then I quit kicad and run the rest of the automation scripts. Obviously, this step should also be automated. However, it will require putting new stuff in kicad-automation-scripts to click all the right menus and buttons in pcbnew UI to generate that file.

I also took a look at KiAuto (the fork of your kicad-automation-scripts) just in case he already added it there, but I didn't see it.

How would you like to see this resolved? I can take a run at it in kicad-automation-scripts, although I am not very familiar at all with the kicad API, I can use what is already there as a model. Or, you expressed interest in replacing nicad-automation-scripts with KiAuto, so maybe that is where the effort should be focused first?

I can also go ahead and make the PR with what I have now if that is helpful.

obra, 2021-09-29:
It *looks* like KiBot can now do .pos files. It may be time for us to jump from kiplot to kibot? (I'm fine with that happening after the initial merge.)

kroesche, 2021-10-02:
I think this can be closed.

closed 2021-10-02 by obra